ThunderFly-aerospace/PX4-Autopilot #2: Implementace měření otáček rotoru

kaklik, 2018-08-14:
U Vírníku je výhodné umět měřit otáčky hlavního rotoru, neboť tato hodnota odpovídá velikosti dostupného vztlaku. V autopilotovi zatím zřejmě není implementován jednotný způsob zacházení s měřenými otáčkami.

Řešením je implementovat čtení I²C čítače a výpočet otáček. Čítač může být například RTC modul [RTC03A](http://mlab.cz/module/RTC03A).

Současná řešení využívají arduino spojené s hallovou sondou:
https://hackaday.io/project/20311-measuring-engine-rpm-with-the-pixhawk

slimonslimon, 2018-11-01:
ve větvi autogyro-mlab je driver mlab_rotorfreq, který publikuje uORB topic rotor_frequency.

- Zatím testovánojen na Linuxu
- Napevno zadrátovaná adresa zařízení

slimonslimon, 2018-11-02:
TODO:
- implementovat:
  - volitelný počet magnetů
- načítat z paramerů: 
  - adresu zařízení
  - počet magnetů
  - interval restartování
  - defaultní pool interval

kaklik, 2018-11-24:
Jo, tohle je už zřejmě v podstatě hotové.  
Jenom by  teď stačilo změnit ty názvy souborů tak, aby to vedlo na použitý typ snímače, nikoli na mlab. a bylo to celé ve složce rpm (snímače otáček). 
Možná se může už odstranit i ten dummy driver.

slimonslimon, 2018-11-24:
to přesunutí a přejmenování mám částečně hotové z pátku,ale zasekl jsem se trochu na verzi ecl -> nerad bych to pushem sotatním rozbil. Pushnu to příští týden (je to v prahe a ja jsem v Trutnově)

slimonslimon, 2018-11-24:
aha, netušilsem že oto/min je žádanější :-D zajím jsme to nazval rotor_frequency (stejně jako message který to rpodukuje, ale tn by se měl menovat sensor_rotor_frequency, mám dojem....

slimonslimon, 2018-11-30:
názvu tachometr jsem se vyhul, pojmenoval jsme to rotor_frequency, což je myslím výstižné, otáčky za minutu jsem přidal.

slimonslimon, 2019-07-17:
Co je se senzorem za problém? 

když jsem dodal driver do čisté PX4 z upstreamu, tak driver nešel přeložit pro px4_fmu-v5, oprava spočívá v zakomentování obsahu funkce read, která se stejně patrně nepoužívá, všichni poslouchají zprávy uORB.

Zbývá odpovědět kam (do jaké větve) tento patch implementovat.

kaklik, 2019-10-07:
Tak momentálně to funguje, zřemě by to bylo v tomhle stavu dobré dostat do upstreamu, abychom se o to už nemuseli víc starat.

slimonslimon, 2019-10-07:
To bude problematický oříšek, si myslím --- což o to čistě driver by tam asi dostat šel, ale pak to napojení na mavlink, to bude horší, tam místo vlasního "žargonu" Mavlinku jsem to přidal přímo do hlavní větve, stydím se, fuj - ale nevím co stím přesně.
Další věc o které netuším jakou představu mají je "další senzor na UAV"... tedy např senzor otáček rotoru - tohle přímé poslání zpráv z driveru do řízení obchází jejích program "sensors" který nějak filtruje senzory a děla failover, když jsou dva senzory na tu samou věc... ale kdysi mi přišlo, že tam mají "každyý tip senzoru" vypsaný..

slimonslimon, 2019-10-07:
a skoro bych to viděl na opačné rvaní věcí do upstreamu -> nejdřív FG (SITL s autogyro nebo Helicoperoou) 
a pak další senzory těchto zařízení

myslím si že by jim to mohlo dávat větší smysl.

kaklik, 2019-10-29:
Podle posledního testu to vypadá, že tenhle driver koliduje s driverem sdp3x_airpeed zapojeným na stejné sběrnici. 
Je možné, že je to vlastnost že na jedné sběrnici nemůže být nic jiného než airspeed, nicméně je to zjištění, které by výhledově chtělo prověřit.

kaklik, 2019-11-14:
Poslední zkoušky ukazují, že kolize souvisela spíš s nějakým speciálním využitím sběrnice I²C 1 v autopilotovi. 

Tohle issue bych ale uzavřel až driver bude v upstreamu.

kaklik, 2020-02-09:
Základní driver už v upstreamu je. Nyní jej stačí doladit tak, aby se stal referenčním driverem. Nicméně podmínky pro zavření issuje jsou splněny. :)

closed 2020-02-09 by kaklik