ThunderFly-aerospace/PX4-Autopilot #6: Simulační vývojové prostředí

kaklik, 2018-09-05:
Pro ověření úprav algoritmů je potřeba mít fungující vývojové prostředí pro testování. 

Zřejmě je vhodné použít některý z [existujících nástrojů](https://dev.px4.io/en/simulation/).

roman-dvorak, 2018-09-05:
Nejdříve otestovat simulační systém gazebo s normálním letadlem (FixedWings) a naučit se pracovat se simulátorem a propojení s QGControl.

Zkusit postupovat podle tohoto návodu
https://dev.px4.io/en/simulation/gazebo.html

kuhnmare, 2018-09-11:
Funkcnost oficialniho px4(v1.8) + Gazebo9 v SITL rezimu s FixedWing letadlem overena. 

Dalsim krokem bude pouzit Thunderfly fork px4.

kuhnmarek, 2018-10-10:
Update:
Jelikoz jsem na internetu nenasel zadny vhodny model jakehokoliv rotoru pro Gazebo, prostudoval jsem jak funguje Yasim. Jeho model rotoru helikoptery je dost pokrocily, ale da se tim dost inspirovat. 

Mezitim jsem vytvoril zaklad pluginu do gazeba (gazebo_rotor_plugin.cpp) a otestoval, ze funguje pro model virniku. Byly tam dost velke problemy s numerickou stabilitou kvuli nekolikanasobnym rotacnim vazbam u rotoru. Uspesne jsem otestoval regulaci otacek rotor nejzakladnejsim proporcionalnim regulatorem s deaktivovanymi pitch, roll aktuatory. Je potreba vyresit zpusob rizeni p. r. os, protoze stavajici "position_kinematic" typ definovany v mavlink_pluginu nelze mixovat s fyzikalni simulaci. Celkove je potreba prozkoumat, jak gazebo pracuje s limity u vazeb, protoze to casto vede k numericke nestabilite.

kaklik, 2018-10-10:
Paráda :) Akorát, kde přesně je ten gazebo_rotor_plugin.cpp ? Já se koukal na [gazebo_rotor_model.cpp](https://github.com/ThunderFly-aerospace/sitl_gazebo/blob/autogyro-sitl/src/gazebo_rotor_model.cpp) A tam jsem žádnou  matematiku, kde bych čekal numerické nestability nenašel. 

Úplně ale nerozumím těmto bodům:

-  Když si testoval regulaci otáček proporcionálním regulátorem, s vypnutými pitch a roll aktuátory, tak jakým akčním členem to regulovalo otáčky?  To pak odpovídá výkonu prerotátoru při roztáčení? 
-  Není mi jasné, proč je potíž se řízením pitch a roll os. Tuším asi, že by to mohlo být tím, že pohyb rotoru simuluješ přes sférické vazby na hlavě.  Ale to bych zatím nedělal a prostě použil pitch a roll úhly rovnou jako pozici rotoru.  Pro zpřesnění simulace se to tam dá později doplnit. Teď to přinese spoustu komplikací. Je mi jasné, že některé jsi už možná vyřešil, ale myslím, že to ještě není konec.

kuhnmarek, 2018-10-11:
Ten gazebo_rotor_model mas spravne. Je to dummy plugin, kam prijdou vypocty tykajici se rotoru. Ty nestability se neobjevuji tam, ale az pri runtime v solveru. Kdyz jsou napr. v definicich gazebo modelu  [https://github.com/ThunderFly-aerospace/sitl_gazebo/blob/autogyro-sitl/models/autogyro/TF-G1.sdf](url) spatne urcite parametry, tak se napr stane, ze vse vypada normalne, akorat ani moment 1000 Nm s rotorem nepohne a vycitani uhlovych rychlosti vraci nesmysly.

Ad 1. Cilem tohoto testu (void GazeboRotorModel::SimpleSpinTest()) je aplikovat moment vypocteny z gazebo pluginu na rotor. Ve finale pujde o rotacni moment z autorotace a vztlak. Neni to napojene pres mavlink, z duvodu Ad 2. Cilova rychlost je hardcoded jen pro demonstraci. 

Ad 2. Puvodne jsem to mel, jak rikas. To se resi zde https://github.com/ThunderFly-aerospace/sitl_gazebo/blob/autogyro-sitl/src/gazebo_mavlink_interface.cpp#L956 . Nicmene michat primo zadane uhly a dynamickou simulaci nevede k dobrym vysledkum. Dochazi napr. k ruznym kolizim typu: V jednom kroku aplikuju moment nebo silu na rotor, coz vyvola nejakou reakci ve vazbach. Pokud v dalsim kroku nastavim natvrdo nejakou pozici, jak se to projevi z hlediska dynamiky na ostatnich castech? Nevim, jestli tam neni nejaka jina chyba, ale aspon takhle si vysvetluju to chovani co jsem dostal. Napr. pri vycitani uhlove rychlosti rotoru to kazdy druhy krok vracelo 0 a kazdy prvni zase dost velke cislo, nezavisle na tom kterou osu jsem takto ridil. Proto chci zkusit pouzit silovy typ regulace uhlu pitch/roll.

slimonslimon, 2020-04-05:
implementací podpory FG toto považuji za vyřízené

closed 2020-04-05 by slimonslimon