ThunderFly-aerospace/PX4-Autopilot #9: Omezení rozsahu setpointů omezí rozsah výstupu kontrolních prvků

kaklik, 2018-10-01:
Zmenšení hodnoty parametrů
FW_MAN_P_MAX,  na  15.0
FW_MAN_R_MAX,  na   5.0
Způsobilo snížení rozsahu výstupních hodnot aktuátorů. Tento efekt je vidět v grafu Actuator Controls.
Je potřeba zjistit, proč se tak děje a zřejmě upravit hodnotu  nějakých  dalších paramerů.

roman-dvorak, 2018-10-14:
Po prozkoumání kódu PX4 toto lze vyřešit změnou (zvětšením) parametrů regulátoru. Ovládání by se následně mělo chovat podstatně citlivěji..

kaklik, 2019-03-26:
Dovedeme určit o kolik se musí zvětšit parametry regulátoru, aby dynamika regulátoru odpovídala původním parametrům s většími rozsahy setpointů? 
Ono by bylo vůbec nejlepší, kdyby změna rozsahu setpointů neměla vazbu na velikost parametrů regulátoru. Myslím že to ani nedává smysl, protože dobře nastavený regulátor na velký rozsah setpointů umí regulovat i menší setpointy. Opačná implikace chápu, že nefunguje.