ThunderFly-aerospace/PX4-Autopilot #14: Vytvoření simulovaného snímače otáček

kaklik, 2018-12-24:
Simulátor Gazebo potřebuje simulované drivery pro poskytování potřebných údají řídícím algoritmům autopilota. 
Existující simulované drivery jsou ve složce [simulator](https://github.com/PX4/Firmware/tree/master/src/modules/simulator).  Na tomto místě je potřeba vytvořit novou kategorii _rpmsim_. A do této složky umístit kód potřebný pro simulaci otáček rotoru.  Potřebné úhlové rychlosti je zřejmě nutné získat z pluginu rotoru.

kaklik, 2018-12-24:
@kuhnmarek  Je možné z pluginu pro rotor získat otáčky rotoru externím driverem?

kaklik, 2020-03-24:
Tohle se zřejmě nyní snáze vyřeší ve FlightGearu..

roman-dvorak, 2020-03-30:
Pro testovací účely bez simulátoru jsem vytvořil jednoduchou aplikace, která vytvoří a pošle jednoduchou uorb RPM zprávu. Aplikace se používá následovně:

```
rpm_simulator <hodnota RPM>
```
Ukazka z shelu:
```
pxh> rpm_simulator 13
INFO  [rpm_simulator] RPM message with 13.0000 RPM was published

pxh> listener rpm
TOPIC: rpm
 rpm_s
	indicated_frequency_rpm: 13.0000
	estimated_accurancy_rpm: 0.1300
```

(neřeší to však původní záměr tohoto issue)

slimonslimon, 2020-03-31:
nabastleno:
7d01b7bdbe911dc1eaa7954637ab194bf7d58aed

kaklik, 2020-03-31:
Pro účely testování ve flightgear zatím vyřešeno.

closed 2020-03-31 by kaklik

slimonslimon, 2021-01-14:
update: Zaktualozovaáno vě větvi: https://github.com/ThunderFly-aerospace/PX4Firmware/tree/FG-rpm