ThunderFly-aerospace/PX4-Autopilot #15: Passthrough FrSky Telemetry Protocol support

kaklik, 2019-05-19:
PX4 nepodporuje [Passthrough FrSky Telemetry Protocol](http://ardupilot.org/copter/docs/common-frsky-passthrough.html) a díky tomu nelze použít https://github.com/yaapu/FrskyTelemetryScript.

slimonslimon, 2021-10-06:
https://docs.px4.io/master/en/peripherals/frsky_telemetry.html

kaklik, 2021-10-06:
> https://docs.px4.io/master/en/peripherals/frsky_telemetry.html

To ale není to samé, jako to co má Ardupilot.

slimonslimon, 2021-10-07:
Aha - to jsem netušil, že ardupilot so po Frsky posílá něc víc. Jakési MavLite zprávy..

slimonslimon, 2021-10-07:
ne že bych se do toho teď vrhal, ale implementace v PX4 je docela čitelná a asi by se dala snadno rozšířit. 

Trochu pohanská otázka je, zda se snažit implementovat "plnotučné" řešení s MavLite (jejich přesnou definici jsem zatím nepotkal), nebo zda si jen neposlat vlastní zprávu, kde by byly věci které nás zajímají navíc proti tomu co se posílá ted což je asi RPM

kaklik, 2021-10-07:
> 
> Trochu pohanská otázka je, zda se snažit implementovat "plnotučné" řešení s MavLite (jejich přesnou definici jsem zatím nepotkal), nebo zda si jen neposlat vlastní zprávu, kde by byly věci které nás zajímají navíc proti tomu co se posílá ted což je asi RPM

Takový přístup by znamenal prepsat ty LUA skripty ve "vysílačce", aby uměl zpracovat zprávy od PX4.    Zatím nevim o nikom kdo by se do tohodle pouštěl. Lidi to řeší tak, že k té vysílačce z venku připojí arduino, které ty zprávy konvertuje.

roman-dvorak, 2021-10-07:
Já mám doma nepushnuté (ale commitlé) rozdělení PX4 pro podporu nového telemetrického protokolu. Tedy ten passtrought. 

Teď rychle vím ještě o skriput pro INav autopilota. Který je možná i kompatibilní s Yaapu. 
Co se týče změny přenosového protokolu (ACCEST/ACCST), tak nevím o tom, že by se měl změnit protokol toho telemetrického port (Sport). Měl by zůstat stále stejný.