ThunderFly-aerospace/PX4-Autopilot #24: Multi vehicle simulation - komunikace

slimonslimon, 2020-04-29:
Pro Start na laně ve FG potřebujeme multi vehicle simulaci, a zároveň simulovat posílání dat z TF-R1 do TF-G1.

V rámci simulace se jeví jako nejednodušší možnost spustit další mavlink stream při spuštění TF-R1 pomocí : ```mavlink start -o 15555 -r 100```, kde 15555 je cílový port kam budou mavlink zprávy posílány.  Vyžaduje to zvýšení možného počtu instanci mavlinku: 
https://github.com/ThunderFly-aerospace/PX4Firmware/blob/flightgear-multi/src/modules/mavlink/mavlink_main.h#L537

Pro TF-G1 je potřeba napsat modul, který by na daném portu seděl a příchozý zprávy (především GLOBAL_POSITION_INT) překládal na uorb zprávu: 
https://github.com/ThunderFly-aerospace/PX4Firmware/blob/flightgear-multi/msg/follow_target.msg
Taková věc je určitě možná, dělá to modul simulátor, při poslechu zpráv ze simulatoru:
https://github.com/ThunderFly-aerospace/PX4Firmware/blob/flightgear-multi/src/modules/simulator/simulator_mavlink.cpp#L692 - vypadá to jako standardní socketová komunikace.

TODO:
- [x] Multi Vehicle simulace
- [ ] Posílání MAVLink zpráv z TF-R1po startu systému
- [ ] Sepsání modulu na příjem zpráv na daném portu.
- [ ] Publikování folow_target uorb
- [ ] Vymyslet, jak tohle dostat ze SITL do reálu.

slimonslimon, 2020-05-02:
k duhému bodu jsem sestavl testovací aplikaci v C která příjmá UDP mavlink zprávy a vypsiuje zprávu o pozici.

slimonslimon, 2020-11-28:
Je start na laně ještě aktuální?

kaklik, 2020-11-28:
Momentálně start na laně aktuální není.  Ale simulace TF-R1 v kombinaci s TF-G1/G2 stále aktuální je, ale nemá to teď moc vysokou prioritu.