ThunderFly-aerospace/PX4-Autopilot #26: Implementation of Ananometer schield into PX4

slimonslimon, 2020-07-07:
Ukolem tohoto tasku je sepsaní poznatků o implementaci podpory https://github.com/mlab-modules/WINDGAUGE03 do PX4.

Destička obsahuje IMU, který má vlastní I2C master, na kterém sedí SPD_3X diff preasure, a magnetometr. včechny tři jednotlivé komponenty by měli být podporované v PX4. (Dokonce se zdá, že IMU má driver napsaný i se slave magnetometrem)

Prvním cílem je vyčítání airspeed pomocí Vernouliho trubice, která bude osazena touto destičkou. 

Jedna z možností je napsat nový driver IMU jednotky, který aktivuje pouze I2C briging, tedy vypne vlastní master a použít původní drivery slave zařízení, především diff preasure senzoru.

Další problím je přepočet Diff preausre na Indicated AirSpeed, která se provádá zde:
https://github.com/ThunderFly-aerospace/PX4Firmware/blob/master/src/lib/airspeed/airspeed.cpp#L81
(asi přidám nový způsob kompenzace, který provede přepočet - tím by pak všechny diff presure senzory mohly být použity pro vernouliho trubici, pokud náš vzorec bude dostatečně univerzální)

slimonslimon, 2020-12-04:
Driver v PX4 pro IMU20948 je pouze pro SPI... zvolena možnost napsat novy driver který jen uděla bridging...

kaklik, 2021-03-09:
Tohle issue je zřejmě redundantní, nebo bylo nahrazeno issue přímo u TFASPDIMU: https://github.com/ThunderFly-aerospace/TFASPDIMU01/issues/3

Závěr myslím je, že bude využit driver, který udělá bridging.

closed 2021-03-09 by kaklik