ThunderFly-aerospace/PX4-Autopilot #27: Úprava odpojovacího kontroléru

roman-dvorak, 2020-07-27:
Odpojovací kontrolér nyní hlídá 4 podmínky, které musí být splněné pro odeslání požadavku na odpojení. Těmi jsou povolení z RC ovladače, minimální airspeed, minimální RPM a přípust motoru nad polovinu. 

Bylo by potřeba přidat podmínku, která bude vírník odpojovat při attitude_setpointu v určitém rozsahu (alespoň maximální hodnotu) a do výpisu vypisovat aktuální nastavení/hodnoty. 

Dalším vylepšením by bylo, aby bylo možné platformu odpojit připojit nějak kontrolovaně elektronicky. Mělo by to být udělat tak, že pokud je vírník DISARMovaný, tak přepnutí vypínače bude přímo ovládat výstup platformy.

kaklik, 2020-12-25:
Poslední úprava odpojovacího kontroleru způsobila, že bylo zrušeno palubní logování tunnel packetu.

kaklik, 2021-01-01:
Nový firmware logování v kombinaci s odpojovacím kontrolerem pravděpodobně opravil.   Potíž ale nastavá s faktem, že zřejmě nestačí, aby celý proces startu končil signálem pro odpojení. 
Na základě posledních pokusů je zde poměrně významný požadavek na přechod do režimu maximálního stoupání a zafixování letového stavu (stabilizace otáček rotoru, stabilizace rychlosti a polohy). Neboť těsně po vypuštění je vírník citlivý na chyby regulátoru i pilotážní chyby, což může mít snadno za následek havárii.

kaklik, 2022-01-02:
Zde je práce také velmi hotová. Zbývá nahradit provizorní mavlink zprávu CAMERA_IMAGE_CAPTURED, některou, která bude akceptovatelná v PX4 pull-requestu.

roman-dvorak, 2022-01-02:
V mavlinku jsem aktuálně nenašel žádnou vhodnou zprávu. 

Navrhoval bych zprávu prozatím nahradit zprávou: https://mavlink.io/en/messages/common.html#DEBUG Po dokončení PR pak vytvořit vlastní zprávu pro tento způsob startu.

kaklik, 2022-01-02:
> V mavlinku jsem aktuálně nenašel žádnou vhodnou zprávu.
> 
> Navrhoval bych zprávu prozatím nahradit zprávou: https://mavlink.io/en/messages/common.html#DEBUG Po dokončení PR pak vytvořit vlastní zprávu pro tento způsob startu.

Ok.. A myslíš, že je pak špatný nápad používat ty zprávy winch actions?

roman-dvorak, 2022-02-07:
Tohle už není aktuální issue, je to řešeno v trochu jiné podobě součástí #32

closed 2022-02-07 by roman-dvorak