ThunderFly-aerospace/PX4-Autopilot #28: Parametry startu konfigurované přes SDkartu

kaklik, 2020-10-27:
Je zřejmě potřeba otestovat, zda se [přes SDkartu](https://dev.px4.io/master/en/concept/system_startup.html?) dají nastavit nové parametry, které jsou potřebné pro správnou funkci indikátorů během vzletu.

roman-dvorak, 2021-11-07:
Asi nevím, jak tohle issue myslíš. Ale nastavení parametrů ze SD karty jsme už několikrát používali. Především v době, kdy jsme ještě neměli vlastní airframe pro TF-G2

kaklik, 2021-11-07:
> Asi nevím, jak tohle issue myslíš. Ale nastavení parametrů ze SD karty jsme už několikrát používali. Především v době, kdy jsme ještě neměli vlastní airframe pro TF-G2

Tohle issue souviselo s tím, že jsme potřebovali během minulých pokusů o vzlet z auta nastavovat parametry, které nebyly součástí parametrů autopilota. 

Nejsem si teď jistý, zda to ještě i teď platí nebo ne. Nicméně jde především o parametry, které pozemní stanici v autě umožní například výpočet optimální rychlosti jízdy a maximální airspeed vzhledem k otáčkám rotoru. 
Bylo k tomu potřeba mít nějaké technické parametry vírníku, které přímo nesouvisely s parametry autopilota.

roman-dvorak, 2021-11-07:
Aha, děkuji. Už vím o co jde. 

Tak to jsem už zjistil, že to nelze udělat. Parametr se nenastaví. Ani vzdáleně přes mavlink to nelze udělat. A vím o těchto možných řešeních:

 * Parametr inicializovat v nějakém out-of-tree modulu. Pak ho lze nastavovat po mavlinku nebo z SD karty. 
 * Na SD kartu uložit soubor, který se pak z SD karty celý stáhne. A bude obsahovat celou konfiguraci startovací platformy.