ThunderFly-aerospace/PX4-Autopilot #29: TFRPM01 could wrongly report negative RPM in some cases

kaklik, 2020-11-29:
Here is a graph showing that situation. 

[image]

It is probably related to an unusual situation like a crash. It should be inspected and the suspicious conditions should be defined to avoid that.

slimonslimon, 2020-12-04:
It needs "imaginal experimet" and code random walk to solve this issue...

roman-dvorak, 2020-12-04:
it occurred, probably, at the moment when wires were broken down.

kaklik, 2021-03-09:
The intensity of occurrence of this bug dramatically increased in combination with TFASPDIMU sensor connected at the same bus as is described in the comment: https://github.com/ThunderFly-aerospace/TF-G2/issues/88#issuecomment-785433829

slimonslimon, 2021-10-06:
I don know how it can occured, but it will be detect as driver error and sensor will be restarted.

roman-dvorak, 2022-12-03:
Negativní hodonty nyní reportovány nejsou. Aktuální nedostatek je, že driver povolí reportování extrémního peaku v kladných hodontách. 

@slimonslimon  narhuje to vyřešit způsobem, že se bude počítat něco jako aktuální derivace a na základě toho se rozhodne, jestli se hodnoty zahodí. 

Napadlo mne, zkusit vyčítat senzor 2x rychle za sebou a hodonty vzájemně porovnat. Jestli peak nevzniká způsobem, že v průběhu vyčítání všech registrů se změní hodnota některého bitu a tím vznikne taková extrémní hodnota. Z datasheetu není zřejmé, že by takováhle záležitost byla řešena.

kaklik, 2022-12-03:
Navrhoval bych zlepšování tohodle issue nechat už na někom dalším. Pro naše použití je to myslím takhle už dostatečné.

closed 2022-12-03 by kaklik