ThunderFly-aerospace/PX4-Autopilot #30: TFRPM01 RPM driver can start on bus without connected sensor board.

roman-dvorak, 2020-11-29:
At driver start-up, the driver should verify that the sensor with the selected address on the given bus really exists. If not, it should exit.

This is the standard way to turn on drivers without having to know the right bus. The standard way is to start the driver with `-X` parameter. This will start the driver in multiple instances (for every external bus) and instances for the bus without sensors will exit

slimonslimon, 2020-12-04:
I know this, but i when i was writing the driver, I didnt found universal I2C "who am I"....

roman-dvorak, 2020-12-04:
Can be it done as:
"Write the configuration register and then read it, if it is written there? If it differs, the sensor is disconnected?"

slimonslimon, 2020-12-05:
I will try improve it. "Onetime, nexttime"

kaklik, 2022-01-13:
The identification procedure for that i2c device could not be implemented. The device does not contain a register, which could be used for reliable identification at startup. All registers are zero after power up and nothing is changed until a pulsed signal appears at the clock input.

closed 2022-01-13 by kaklik

roman-dvorak, 2022-01-13:
I solved this by writing to free RAM registers, which will not affect anything. They will then verify their value. It is done on multiple registers, where it is unlikely that it will work fully with another IO. However, it could be reset in some way. But this is practically only used during initialization. So it wouldn't have to be a problem. 

 * But the main result is that the driver cannot run on a non-existent device. 
 * And I don't expect to start the driver on random buses with a random I2C address.