ThunderFly-aerospace/PX4-Autopilot #32: Autogyro takeoff mode

roman-dvorak, 2021-01-12:
After our ThunderFly devcall with @kaklik and @slimonslimon we agreed on the following implementation of takeoff mode for the autogyro. 

Why:
- FW takeoff (from runway) sequence is different from autogyro sequence 
- We want takeoff from some moving platform (car roof or running man)

## How to do takeoff with an autogyro

1. At the beginning, it is assumed that someone pre-rotates the rotor to the minimum speed for the prerotator. This conditions will be set in some parameter
2. Then it starts prerotator and it is waiting for obtaining flight RPM of rotor. (Prerotation can be done by some prerotator driver or with forwarding movement). Rotor (pitch) is at some maximal setpoint value.
3. Now, autogyro is in flight parameters. Until this point, integrators are inactive. Now, release command is transmitted (it can be mavlink command or some sound signal for running man) 
4. Stage of climbing, it is similar to FW. Controlled by TECS with limited setpoints range. 

## Implementation
Because VTOL autogyro is not assumed (VTOL in the sense that vertical and horizontal flight is controlled by different actuators) 

## Testing:
Because it is better to perform the first tests in the simulator, @kaklik will prepare a modification of the TF-G2 autogyro model it will contain dummy (very heavy) weights that is possible to switch off with some keyboard command. It will simulate takeoff from running man. (more edails in own issue) 

Then we try to perform some real test flights form a hand or from the car roof. Here we are missing a piece of SW and (HW)[https://github.com/ThunderFly-aerospace/TensegrityPlatform], which will disconnect the rotor from the roof from the mavlink message.

kaklik, 2021-02-02:
The issue has been almost solved. There is needed only protection of rotor overspeed which could happen in the case where the autogyro does not been released from platform promptly and platform speed increases.

The protective algorithm should lean forward the rotor and interrupt the takeoff if applicable.

kaklik, 2022-01-02:
The flight-tested performance of the current implementation of the takeoff procedure is perfect! It could also be documented by several flight logs and video recordings. 
Therefore merging that to upstream is mandatory, until there are no changes that affect the functionality. :)

kaklik, 2022-01-02:
I am closing this issue, as the main discussion already moved to PX4 upstream PR: https://github.com/PX4/PX4-Autopilot/pull/18582

kaklik, 2022-01-13:
Nějak se zadrho přijetí pull-requestu, takže to zřejmě bude potřebovat ještě nějakou péči.