ThunderFly-aerospace/PX4-Autopilot #35: Vertical wind velocity estimation

kaklik, 2021-08-13:
Při diskusi s @povik  jsme přišli na to, že [TECS regulátor](https://docs.px4.io/master/en/flight_stack/controller_diagrams.html#total-energy-control-system-tecs). Zřejmě obsahuje hodnotu kompenzace vertikální rychlosti.  Hodnotě odpovídá integrace ve větvi označené jako Throttle compensation na následujícím diagramu.

[image]

Tato integrace se počítá na [tomto řádku](https://github.com/PX4/PX4-Autopilot/blob/2cf66a5d8f96a0d129c544580c07f3b538f3fabc/src/lib/tecs/TECS.cpp#L323) TECS regulátoru. 

V případě vyvedení této hodnoty do uORB zprávy bychom mohli 

- Určit rychlost vertikálního proudění v prostoru
- Změřit rychlost opadání vírníku během ustáleného letu (aerodynamickou účinnost).

povik, 2021-08-16:
Nevím, do jaké míry diagram odpovídá kódu v TECS.cpp, protože v kódu do integrace vstupuje (po škálování)  `_STE_rate_error`, ale podle diagramu bych čekal spíš  `_STE_error`. Nic to ale na té původní úvaze, ze které vyšlo, že "throttle compensation" obsahuje zajímavou informaci, nemění.

kaklik, 2022-05-03:
Tak jsem u tohoto issue chtěl přidat logování throttle_compensation.  A když jsem to začal dělat, tak jsem si všiml, že [už existuje uOrb zpráva](https://github.com/PX4/PX4-Autopilot/blob/138772386f0f0cb462c0b6972f442b94aac5d038/src/modules
/fw_pos_control_l1/FixedwingPositionControl.cpp#L538), která stejnou hodnotu obsahuje. 

Při posledním letu z platformy je v ní tohle: 

[image]

Vzniká z toho tak graf o kterém moc nevím co přesně znamená. Hlavně mi jsou podivné ty nespojistosti. Část toho grafu tak zřejmě obsahuje informaci o vertikálním proudění, ale zároveň chybí vazba, která by umožnila korektní interpretaci.

kaklik, 2022-05-04:
Tak nespojistosti viditelné v grafu lze vysvětlit chybějícími daty. V režimu _Stabilized_  TECS vůbec nefunguje a když se mód letu přepne ze _Stabilized_, do _Position_, tak se ta naintegrovaná hodnota vynuluje. Takže validní hodnoty jsou jen v _Position_, _Loiter_ a částečně v _Mission_ těsně po startu.  Tam to je ale ještě pořád takeoff, takže záleží na tom, jak s tou naintegrovanou hodnotou zachází kód od @roman-dvorak. Aktuální mód je poznamenán ve zprávě [vehicle_status/nav_state](https://github.com/PX4/PX4-Autopilot/blob/138772386f0f0cb462c0b6972f442b94aac5d038/msg/vehicle_status.msg?#L31)

Pokud se ale provede průzkum delšího letu, kde se mezi módy nepřechází. Tak hodnota throttle_integ od nějaké fáze letu spojitě roste. Zde je například záznam letu ze Silvestrovského meření prachu. Je vidět, že ke kontinuálnímu nárůstu throttle_integ dříve než v maximální výšce letu.  Zároveň ale začátek stoupání throttle_integ nelze jednoduše vysvětlit například poklesem napětí akumulátoru (pokud regulátor nemá nějaký zvláštní treshold v účinnosti a napájecím napětí).

[image]

[image]

[image]

kaklik, 2022-05-06:
Na letu provedeném 16.9.2021 jsem zjistil, že hodnota throttle_integ  může být i záporná

[image]

kaklik, 2022-05-07:
> Nevím, do jaké míry diagram odpovídá kódu v TECS.cpp, protože v kódu do integrace vstupuje (po škálování) `_STE_rate_error`, ale podle diagramu bych čekal spíš `_STE_error`. 

@roman-dvorak  našel [tenhle pull-request](https://github.com/PX4/PX4-Autopilot/pull/16362), kde zřejmě došlo k těto zásadním změnám, které jsou v rozporu s dokumentací (která od té doby nebyla aktualizována).

kaklik, 2022-05-12:
Tak jsem [překreslil TECS throttle controller](https://github.com/PX4/PX4-user_guide/pull/1880#issuecomment-1125199227) podle aktuálního kódu.

kaklik, 2022-05-14:
Ono to řešení extrakce účinnosti letu a měření externího vertikálního proudění je vlastně celkem přímočarý. Jde o to zjistit, jaká je ztráta (total energy rate) v čase samotným letem versus  total energy rate, která přibývá motorem.  Ten tok energie z motoru by se asi dal považovat za lineární funkci throttle. Jen [chybí konstanta, kterou se throttle přepočítá na energy rate](https://github.com/ThunderFly-aerospace/TF-G2/issues/43#issuecomment-1126716682). (tím by zřejmě dokonce šlo měřit absolutní účinnost motoru).

Pak následně, když se bude měnit total energy a nebude za to moct throttle. Tak je to externí vliv, který zřejmě nemůže být jiný než vertikální proudění.  Může se to ale možná zkomplikovat ještě tím, že převod mezi kinetickou energií a potenciální energií má ztráty.   Ztráty by ale zas v nějakém rozumném rozsahu měly být funkcí kvadrátu rychlosti. 

Tím by řešení uvnitř TECS potřebovalo během letu stále fitovat několik konstant, které by byly na počátku zinicializované parametrem. Jako je to například teď u převodu mezi [throttle a energy rate](https://github.com/PX4/PX4-Autopilot/blob/3fe4c6e3f56d68417918454188a3f968c1e5ccd6/src/lib/tecs/TECS.cpp#L289).