ThunderFly-aerospace/PX4-Autopilot #36: Signing MAVLink 2 messages support

kaklik, 2021-09-27:
Pro rozsáhlejší mise bychom potřebovali zabezpečit autentizaci příkazů posílaných z pozemíní stanice do autopilotu (uplink). 
[Specifikace Mavlink 2 takovou záležitost obsahuje](https://mavlink.io/en/guide/message_signing.html). Zatím ji však zřejmě podporuje pouze Adrdupilot v kombinaci s [Mission plannerem](https://ardupilot.org/planner/docs/common-MAVLink2-signing.html).

Ohledně podpory v PX4 proběhly první [pokusy o implementaci již v roce 2017](https://github.com/mavlink/MAVSDK-Java/issues/35).  Aktuálně snaha o implementaci v PX4 zřejmě [probíhá v tomto pull-requestu](https://github.com/PX4/PX4-Autopilot/pull/17084).

roman-dvorak, 2021-09-28:
Netuším, jak se QGC chová, když přijme podepsaný paket. Ale pokud by QGC sloužil pouze pro zobrazování polohy a navigaci by řešil nějaký python, tak to ani nevadí, že to QGC neumí. Implementaci do QGC považuji za nejsložitější k vyřešení celého problému.

kaklik, 2021-09-28:
> Netuším, jak se QGC chová, když přijme podepsaný paket. Ale pokud by QGC sloužil pouze pro zobrazování polohy a navigaci by řešil nějaký python, tak to ani nevadí, že to QGC neumí. Implementaci do QGC považuji za nejsložitější k vyřešení celého problému.

Jde mi o to, že potřebujeme nějaký flexibilní způsob jak posílat podepsané příkazy vírníku. Nemůžeme to nechat pouze na python skriptu. Neboť budeme potřebovat umět vyřešit situaci, kdy python skript selže.   O přijímání podepsaných packetů do QGC jsem ani neuvažoval. Předpokládám, že packety budou podepsané pouze směrem k vírníku.