ThunderFly-aerospace/PX4-Autopilot #38: ag_automatic_takeoff: Lockdown if the automatic takeoff is interrupted

roman-dvorak, 2022-03-07:
V případě, že neproběhne celý start, je potřeba vírník uzemnit. Aby nemohlo dojít k nějakým nenadálým a neošetřeným problémům. Lockdown způsobí, že autopilot je potřeba zrestartovat pro provedení startu/nebo vůbec naarmování. 

Lockdown se provede takto: 
https://github.com/PX4/PX4-Autopilot/blob/11f617ca9be36684388c2d1c2a2e1f6698597e44/src/modules/commander/Commander.cpp#L415-L418

Parametr s hodnotou 2 provede lockdown (vypnutí motorů), hodnota 1 je failsafe a mělo by to např. vystřelit i padák. My chceme nejspíše 2. 

Příkaz je zpracován zde:
https://github.com/PX4/PX4-Autopilot/blob/11f617ca9be36684388c2d1c2a2e1f6698597e44/src/modules/commander/Commander.cpp#L1024-L1040

Je potřeba se rozhodnout, za jakého stavu se má start přerušit/ukončit/uzemnit. 

Navrhuji následující podmínky:
 - [ ] Bylo dosaženo kroku 1 (viz obrázek níže) a není větší než krok 3 (To by už měl být ve vzduchu)
 - [ ] Otáčky rotoru jsou menší než polovina RPM potřebná pro daný krok
 - [ ] Výška AGL není větší než (cca 2m?)  (pro případ výpadku měření RPM a utrhnutí z platformy)
 - [ ] Disarmování (tohle aktuálně nevím, jak tam implementovat)
 

[image]

kaklik, 2022-03-11:
Já jsem si tedy původně myslel, že to ukončení startu bude hlavně na tlačítko.. Tj. půjde ovládat posádkou auta. 

Automaticky se to může taktéž provést, ale to je myslím složitější a náročnější krok pro další iteraci.  Jak to myslel @slimonslimon ?

kaklik, 2022-07-30:
Po včerejších pokusech se ukazuje že: 

- autopilota lze po použití kill-switche opět naarmovat bez restartu
- Nemá zřejmě smysl implementovat přerušení startu s kontrolou na úrovni auta, neboť vykonání otevření platformy má oproti vytvoření příkazu k vypuštění vírníkem zanedbatelné spoždění. Není zde tak prakticky žádný prostor, kdy by implementace kontroly na úrovni auta, měla šanci situaci zachránit. 

Ve všech případech by ale zmáčknutí kill-switche mělo zabránit otevření zámku platformy, pokud je to ještě aplikovatené a uvedení rotoru do předkloněného stavu.

kaklik, 2024-06-08:
Tohle se teď v aktuální implementaci zřejmě chová jinak. Kill-switch už nyní nezpůsobuje nutnost restartu autopilota.