ThunderFly-aerospace/PX4-Autopilot #42: Turbine Engine in JSBSim model not working

fahadmaqbool16, 2022-08-23:
Hi, 
I am trying to model my aircraft through JSBSim by modifying the given example of rascal. I have built the aero model and engine file (through aeromatic) and replaced it in the set file. The parameters in these generated files have been thoroughly verified. I have also placed the geometry file which produces the geometry correctly. The aircraft uses the turbine engine modeling [P550 pro](https://www.jetcat.de/en/productdetails/produkte/jetcat/produkte/Professionell/P550%20PRO%20-%20GL). This turbine engine does not work but the electric engine given in rascal JSBSim model works fine. 

When running the sitl simulation in flightgear, the aircraft is unable to move at all.  Please tell the possible sources for this issue. 
The engine file is shown below:

```
<turbine_engine name="Engine">
  <milthrust>   110.2 </milthrust>
  <bypassratio>     0.0 </bypassratio>
  <tsfc>            0.5 </tsfc>
  <bleed>           0.03</bleed>
  <idlen1>          30.0 </idlen1>
  <idlen2>         60.0 </idlen2>
  <maxn1>           100.0 </maxn1>
  <maxn2>         100.0 </maxn2>
  <augmented>         0 </augmented>
  <injected>          0 </injected>

  <function name="IdleThrust">
   <table>
    <independentVar lookup="row">velocities/mach</independentVar>
    <independentVar lookup="column">atmosphere/density-altitude</independentVar>
    <tableData>
           0         5000      10000     15000     20000
     0.15  0.383427  0.317857  0.261733  0.213963  0.173822 
     0.30  0.410490  0.340561  0.280804  0.229765  0.186900  
     0.50  0.473698  0.393235  0.324396  0.265729  0.216143  
     0.65  0.533455  0.443184  0.365627  0.299512  0.243751  
     0.85  0.581769  0.533273  0.444818  0.364355  0.296243    
    </tableData>
   </table>
  </function>

  <function name="MilThrust">
   <table>
    <independentVar lookup="row">velocities/mach</independentVar>
    <independentVar lookup="column">atmosphere/density-altitude</independentVar>
    <tableData>
           0         5000      10000     15000     20000
     0.15  0.928143  0.770486  0.635169  0.519833  0.422478 
     0.30  0.909253  0.754502  0.621910  0.508935  0.413578  
     0.50  0.958112  0.794824  0.654967  0.535816  0.435192
     0.65  1.053106  0.873471  0.719629  0.588490  0.477876  
     0.85  1.160269  1.038939  0.868022  0.709820  0.576139   
    </tableData>
   </table>
  </function>

</turbine_engine>
```

The engine file is included in the aero model as follows:

```
<propulsion>
        <engine file="Old_HSTD_Engine">
            <location unit="IN">
                <x> 102 </x>
                <y> 0 </y>
                <z> 0 </z>
            </location>
            <orient unit="DEG">
                <roll> 0.0 </roll>
                <pitch> 0 </pitch>
                <yaw> 0 </yaw>
            </orient>
            <feed>0</feed>
            <thruster file="Direct">
                <location unit="IN">
                    <x> 1 </x>
                    <y> 0 </y>
                    <z> 0 </z>
                </location>
                <orient unit="DEG">
                    <roll> 0.0 </roll>
                    <pitch> 0.0 </pitch>
                    <yaw> 0.0 </yaw>
                </orient>
                <p_factor>1.0</p_factor>
            </thruster>
        </engine>
        <tank type="FUEL">    <!-- Tank number 0 --> 
            <location unit="IN">
                <x> 55.90 </x>
                <y> 0 </y>
                <z> 0 </z>
            </location>
            <capacity unit="LBS"> 14 </capacity>
            <contents unit="LBS"> 14 </contents>
        </tank>
    </propulsion>

```

kaklik, 2022-08-23:
Hello, we are not specialists in turbine engines. We use electric motors primarily in [YASim](https://wiki.flightgear.org/YASim) models.  Therefore, we cannot help with that. I suggest an opening discussion on some JSBsim-focused forums or using the YASim Jet engine.

kaklik, 2022-08-24:
Closing this as duplicate to: https://github.com/ThunderFly-aerospace/PX4-FlightGear-Bridge/issues/29